LucasAndrad/trocador-calor-pi1-turma-A #1: Treinamentos

LucasAndrad, 2018-08-30:
Organizar treinamentos para ensinar Git e Js

## Git:
### Atividades:
**Atividade 1:** Fazer uma dinâmica com um pequeno projeto em HTML, CSS e JS usando, usando o mesmo projeto.<br>
**Tempo estimado:** 2 semanas (todo sábado ou domingo de manhã)

---

**Atividade 2:**
**Tempo estimado**: 1 semana

---

## JS:
### Atividades:
**Atividade 1:** Criar uma modal sozinho, sem usar nenhuma lib, apenas CSS, HTML e JS<br>
**Tempo estimado:** 1 semana

---

**Atividade 2:** Criar um gráfico simulando os últimos 30 registros de temperatura do trocador de calor<br>
**Tempo estimado:** 1 semana

---

**Atividade 3:** ...
**Tempo estimado:** 1 semana

LucasAndrad, 2018-08-30:
## Passo a passo
### Atividades GIT
- Ter linux
- Instalar o git
- Baixar o editor de texto Visual Studio Code

### Atividades JS
- Baixar o editor de texto Visual Studio Code
- Fazer um clone (`git clone`) do [repositório desse link aqui](https://github.com/LucasAndrad/learn-git-js-electron) e criar sua branch para começar a trabalhar

### Atividades Electron JS
- Criar um repositório para você
- Instalar o framework
- Subir tudo nesse novo repositório
- Deixar um comentário nessa issue com o link do seu repositório

guilhermesiqueira, 2018-09-06:
- Repositório que realizarei as atividades de treinamento relativas ao electron: https://github.com/guilhermesiqueira/learn-electron

gabrielademoraes, 2018-09-06:
- Repositório das atividades de treinamento do electron: https://github.com/gabrielademoraes/PI-electron

closed 2018-09-28 by LucasAndrad