LucasAndrad/trocador-calor-pi1-turma-A #5: Mostrar o valor dos mocks na imagem

LucasAndrad, 2018-09-11:
Vamos mostrar o valor dos mocks na imagem do trocador (issue #4)

LucasAndrad, 2018-10-08:
@andrelucax terminei a revisão do component display data que você fez.
Ajustes:
Não vamos usar Jquery, então você fazer uma pequena refatoração na função que fez para consumir o json.
**Dica:** como consumir arquivo JSON? [Veja a segunda resposta desse link](https://stackoverflow.com/questions/19706046/how-to-read-an-external-local-json-file-in-javascript)

Quando passei seu trabalho pro projeto os dados não são mais carregados na tela. Mas ele continua chamando a função.

Tenta centralizar a div do trocador no centro dessa div azul, e depois pode remover o fundo azul, é pra vc ver qual o espaço que o trocador vai ficar.

PS: não vamos usar Jquery, só o JS mais puro mesmo.
Crie uma nova branch a partir da dev

andrelucax, 2018-10-08:
@LucasAndrad vi que conseguiu consumir os dados do mock. resolvendo a issue. criarei uma nova issue referente a centralizacao da imagem pois ela nao faz mais sentido aqui

andrelucax, 2018-10-09:
Como talvez nem iremos usar JSON recomendo deixar com Jquery por enquanto. Estarei criando PR com imagem centralizada e nomes mudados no css.

closed 2018-10-21 by andrelucax