LucasAndrad/trocador-calor-pi1-turma-A #57: Ajustar simulador de acordo os dados que são usados do arduino

LucasAndrad, 2018-10-31:
Adicionar o FLUXO QUENTE e o FLUXO FRIO na saída

SENSOR =  28 D C7 84 1B 13 1 9C
28.62
FLUXO QUENTE: 0.00 L/min
FLUXO FRIO: 0.00 L/min
SENSOR =  28 2F 99 97 1B 13 1 C3
28.25
FLUXO QUENTE: 0.00 L/min
FLUXO FRIO: 0.00 L/min
FLUXO QUENTE: 0.00 L/min
FLUXO FRIO: 0.00 L/min
SENSOR =  28 AA AA 3E 13 13 2 F5
28.31
FLUXO QUENTE: 0.00 L/min
FLUXO FRIO: 0.00 L/min
SENSOR =  28 AA 81 EE 12 13 2 D7
28.37
FLUXO QUENTE: 0.00 L/min
FLUXO FRIO: 0.00 L/min
SENSOR =  28 59 DD 92 1B 13 1 9E
28.25
FLUXO QUENTE: 0.00 L/min
FLUXO FRIO: 0.00 L/min
SENSOR =  28 D C7 84 1B 13 1 9C
28.69
FLUXO QUENTE: 0.00 L/min
FLUXO FRIO: 0.00 L/min

andrelucax, 2018-11-02:
Creio que podemos ajustar os valores de chegada com a equipe de sensores para nao tem que fazer nenhum tratamento muito especifico, pois do jeito que esta teriamos que tirar um valor float do meio de uma string. Ou acha que devemos deixar assim ? @LucasAndrad

andrelucax, 2018-11-02:
Caso chegue:
```
SENSOR = 28 D C7 84 1B 13 1 9C
28.69
0.00
0.00
```
Poderiamos retirar o valor do float direto, que nem fazemos com o valor da temperatura

LucasAndrad, 2018-11-02:
Podemos mesmo. Vou passar isso pro pessoal de eletrônica.

andrelucax, 2018-11-03:
Alterado no PR https://github.com/LucasAndrad/trocador-calor-pi1-turma-A/pull/58

closed 2018-11-03 by andrelucax